The-OpenROAD-Project-Attic/FastRoute #3: Problems building FastRoute

oharboe, 2019-08-01:
I have a build setup for our project that uses a Dockerfile set up under Ubuntu.

I'd like to add FastRoute and TritonRoute to this existing setup, but when I try to build FastRoute with the attached Dockerfile, it fails as below.

To reproduce, unzip the atatched openroad.zip file and run the Docker command below:

```
# docker build -t openroad openroad/
[deleted]
[ 99%] Building CXX object third_party/rsyn/CMakeFiles/rsyn.dir/src/script/Script.yy.cc.o
[100%] Linking CXX executable bin/rsyn
/usr/bin/ld: /FastRoute4-lefdef/third_party/rsyn/lib/linux/libliberty.a(token.o): relocation R_X86_64_32S against undefined symbol `token_q_buf' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /FastRoute4-lefdef/third_party/rsyn/lib/linux/libliberty.a(mymalloc.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /FastRoute4-lefdef/third_party/rsyn/lib/linux/libliberty.a(PI.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /FastRoute4-lefdef/third_party/rsyn/lib/linux/libliberty.a(liberty_parser.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIC
[deleted]
```

[openroad.zip](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/files/3456301/openroad.zip)

vvbandeira, 2019-08-01:
Hi @oharboe,

We do not officially support builds on Ubuntu (although it is possible to build).

Please consider using the same docker file from the alpha-release repository (known to work) [here](https://github.com/The-OpenROAD-Project/alpha-release/blob/master/build/docker/fastroute/Dockerfile).

But taking a quick look in your file, our makefile does not have an "install" rule, so the last command will fail. Also, check the version of the programs/libs. For example, FastRoute4-lefdef is known to build with boost v1.58 and v1.63 and I do not know which is the default for Ubuntu 18.04 when you run apt install libboost-all-dev.

Regards,

Vitor

oharboe, 2019-08-01:
I have a different context that I want to try FRlefdef in, and that
environment uses Ubuntu. The supported Dockerfile. CentOS 6, served us
well, but it is very old(8 years) and hard to combine with other projects.

> Hi @oharboe <https://github.com/oharboe>,
>
> We do not officially support builds on Ubuntu (although it is possible to
> build).
>
> Please consider using the same docker file from the alpha-release
> repository (known to work) here
That won't work as I need to tinker(debug/tweak, possibly file a pull
request) with FRlefdef, so I need to set up a tinkering environment.

> But taking a quick look in your file, our makefile does not have an
> "install" rule, so the last command will fail.
It doesn't get as far as the "make install", it fails during "make
PARALLEL=8"

> Also, check the version of the programs/libs. For example,
> FastRoute4-lefdef is known to build with boost v1.58 and v1.63 and I do not
> know which is the default for Ubuntu 18.04 when you run apt install
> libboost-all-dev.
Ubuntu 18.04 should be using boost 1.65.1, not far from 1.63 which has been
tested with FastRoute, so I wouldn't suspect a versioning problem without
more investigation.

I wonder if it is a problem with static vs. dynamic boost libraries

> Regards,
>
> Vitor

vvbandeira, 2019-08-01:
On 01/08/2019, Øyvind Harboe wrote:
> I have a different context that I want to try FRlefdef in, and that
> environment uses Ubuntu. The supported Dockerfile. CentOS 6, served us
> well, but it is very old(8 years) and hard to combine with other projects.
> 
> On Thu, Aug 1, 2019 at 1:36 PM Vitor Bandeira <notifications@github.com>
> wrote:
> 
> > Hi @oharboe <https://github.com/oharboe>,
> >
> > We do not officially support builds on Ubuntu (although it is possible to
> > build).
> >
> > Please consider using the same docker file from the alpha-release
> > repository (known to work) here
> > <https://github.com/The-OpenROAD-Project/alpha-release/blob/master/build/
> docker/fastroute/Dockerfile>
> > .
> >
> That won't work as I need to tinker(debug/tweak, possibly file a pull
> request) with FRlefdef, so I need to set up a tinkering environment.

Hum, I understand your position. But I think it is possible as I am able to
build on Ubuntu 16.04.

> > But taking a quick look in your file, our makefile does not have an
> > "install" rule, so the last command will fail.
> >
> It doesn't get as far as the "make install", it fails during "make
> PARALLEL=8"

I know, but just a heads up.

> Ubuntu 18.04 should be using boost 1.65.1, not far from 1.63 which has been
> tested with FastRoute, so I wouldn't suspect a versioning problem without
> more investigation.
>
> I wonder if it is a problem [...]

I don't believe its a problem with static vs dynamic (both approaches work with
boost 1.58 and 1.63). If you can install and test with either of these versions
(keeping all other packages fixed) we could try to find a solution from there.

Regards,

-- 
Vitor Bandeira

oharboe, 2019-08-01:
As a test, I changed my Dockerfile to use Ubuntu 16.04, it has boost 1.58. It builds. Dockerfile attached for reference.
[openroad-ubuntu-16.04.zip](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/files/3457323/openroad-ubuntu-16.04.zip)

oharboe, 2019-08-01:
I've modified my setup to use Ubuntu 18.04 and build boost 1.63, it still fails with the same error message

Using the attached Dockerfile:

```
 $ docker build -t openroad openroad/
[deleted]
-- Boost version: 1.63.0
[deleted]
[100%] Linking CXX executable bin/rsyn

/usr/bin/ld: /FastRoute4-lefdef/third_party/rsyn/lib/linux/libliberty.a(token.o): relocation R_X86_64_32S against undefined symbol `token_q_buf' can not be used when making a PIE object; recompile with -fPIC
```

oharboe, 2019-08-01:
I checked: boost 1.58 failed to compile under Ubuntu 18.04.

So it would appear that the problem is not in boost as we have the same errors with two versions of boost one of them, 1.63, is known to work.

oharboe, 2019-08-02:
@vvbandeira Can we reopen this issue?

Although OpenROAD currently is only known to work on CentOS 6 currently, surely that can't be the ambition of OpenROAD.

I would suggest that this issue falls into an open issue category where you welcome pull requests.

vvbandeira, 2019-08-14:
Hi @oharboe, can you check if the new version in master is buildable with our setup. We removed static libs (.a) and are compiling from source.

oharboe, 2019-08-14:
@vvbandeira It's easier to fix code than documentation: the makefile needs to be updated to check if the submodules needs to be initialized and /or updated.

vvbandeira, 2019-08-18:
@oharboe, I believe the added test in `9608000827f2ba5fb2b6d4b75a2c20c00f47ad27` solves this. 

Can you test with the new version on the master branch (`dd29747029fe0e4837da8696153998d5dd91aa89`)? 

Thanks for your feedback.

(ps. please review issue #4, as I think they are closely related)

oharboe, 2019-08-19:
@vvbandeira  Yes: works now, even if you clone without doing it recursively.

closed 2019-08-19 by oharboe