The-OpenROAD-Project-Attic/FastRoute #4: FastRoute4-lefdef has binary files in git source control that are not compatible with the latest gcc

oharboe, 2019-08-02:
If you try to build FastRoute4-lefdef with the latest gcc, then it will fail with strange error messages.

This log explains it: https://nanxiao.me/en/gccs-enable-enable-default-pie-option-make-you-stuck-at-relocation-r_x86_64_32s-against-error/

The solution is fairly simple: 

1. remove the binary files from the git repository https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/tree/master/third_party/rsyn/lib/
2. update the build scripts for FastRoute4-lefdef to build the third party libraries.

vvbandeira, 2019-08-06:
Hi @oharboe, thanks for opening these issues regarding build, 

A more sane/easier approach would be to add these repositories as submodules in third_party and remove the .a files. I believe not much change is needed in the CMake file. 

We are with other deadlines ahead, but I will find some time (maybe in the near future) to address all these build problems.

Regards.

oharboe, 2019-08-07:
@vvbandeira  Thanks! I noticed that third_party/flute is quite different from the flute master branch, so source code changes to FastRoute4-lefdef will be required there.

I forget which, but there's another module in OpenROAD that uses submodules in this way, presumably to solve the same problem.

vvbandeira, 2019-08-14:
Hi @oharboe, can you check if the new version in master is buildable with our setup. We removed static libs (.a) and are compiling from source.

oharboe, 2019-08-19:
Fixed: this was indeed a problem with incompatibilities across gcc versions that's now solved as everything is built from scratch.

closed 2019-08-19 by oharboe