The-OpenROAD-Project-Attic/FastRoute #5: Crash in FastRoute4-lefdef

oharboe, 2019-08-06:
I've got a testcase that crashes with the alpah-release docker iamge.

I've built FRlefdef for Ubuntu 18 and I ran it in the debugger, which gives me the trace below:

```
# gdb --args ~/FastRoute4-lefdef/FRlefdef --no-gui --script `pwd`/fastroute.rsyn
[deleted]
LV Time: 14.811055 sec
updateType 1
iteration 1, enlarge 9, costheight 34, threshold 0 via cost 0 
log_coef 0.223050, healingTrigger 0 cost_step 30 L 0 cost_type 1 updatetype 1
initial grid wrong y1 x1 [0 0] , net start [10 0] routelen 305
 checking failed 0

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x44000001e14) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x44000001e14) at malloc.c:3103
#1  0x0000555555cf587b in FastRoute::reInitTree(int) ()
#2  0x0000555555cfa598 in FastRoute::mazeRouteMSMD(int, int, float, int, int, char, int) ()
#3  0x0000555555ccd6dd in FastRoute::FT::run(std::vector<FastRoute::NET, std::allocator<FastRoute::NET> >&) ()
#4  0x0000555555a74089 in Rsyn::FastRouteProcess::run(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator> const&) ()
#5  0x0000555555a4e279 in Rsyn::Session::runProcess(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator> const&) ()
#6  0x0000555555a50069 in std::_Function_handler<void (ScriptParsing::Command const&), Rsyn::Session::registerDefaultCommands()::{lambda(ScriptParsing::Command const&)#4}>::_M_invoke(std::_Any_data const&, ScriptParsing::Command const&) ()
#7  0x0000555555a5bc57 in ScriptParsing::CommandManager::evaluateCommand(ScriptParsing::ParsedCommand const&) ()
#8  0x0000555555a50dfb in Rsyn::Session::evaluateFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#9  0x0000555555a61c75 in Rsyn::Shell::runScript(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
#10 0x00005555559036e7 in main ()
```

oharboe, 2019-08-06:
Here's a Valgrind run, which has turned up a number of things to look at.

```
$ valgrind  ~/FastRoute4-lefdef/FRlefdef --no-gui --script `pwd`/fastroute.rsyn
==23521== Memcheck, a memory error detector
==23521== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==23521== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==23521== Command: /home/oyvind/FastRoute4-lefdef/FRlefdef --no-gui --script /home/oyvind/aptos/verilog/alpha-release/flow/test/fastroute.rsyn
==23521== 
==23521== Warning: set address range perms: large range [0xdd9000, 0x1415e000) (defined)

                             _____                  
                            |  __ \                 
                            | |__) |___ _   _ _ __  
                            |  _  // __| | | | '_ \ 
                            | | \ \\__ \ |_| | | | |
                            |_|  \_\___/\__, |_| |_|
                                        __/ /       
                                       |___/        

-------------------------
3rd Party License Notices
-------------------------
1) FLUTE is a software developed by Dr. Chris C. N. Chu at Iowa State University (http://class.ee.iastate.edu/cnchu/)

Running generic reader...
Parsing LEF files...
WARNING: Unsupported INOUT direction in data pin. VDD. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VSS. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VDD. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VSS. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VDD. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VSS. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VDD. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VSS. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VDD. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
WARNING: Unsupported INOUT direction in data pin. VSS. Pin direction is replaced to OUTPUT [LEF CONTROL PARSER]
Parsing LEF files... Done (runtime: 1.49394 seconds memory: +31 MB)
Parsing DEF files...
Parsing DEF files... Done (runtime: 13.71 seconds memory: +60 MB)
Populating the design...
Loaded Design
	Name: LEArrayScanChain
	#Moveable Insts: 14886
	#Fixed Insts: 2457
Populating the design... Done (runtime: 4.36913 seconds memory: +85 MB)
Running generic reader... Done (runtime: 19.6731 seconds memory: +177 MB)
WARNING: Service 'rsyn.graphics' is already running.
FastRoute......

----------------
Params: 
**** Output file: ./route.guide
**** Capacity adjustment: 0
**** Max routing layer: -1

----------------
Initing grid...
Initing grid... Done!
Setting capacities...
Setting capacities... Done!
Setting layer direction...
layerOrientation = 0
layerOrientation = 0
Setting layer direction... Done!
Setting spacings and widths...
Setting spacings and widths... Done!
Initing nets...
Initing nets... Done!
Adjusting grid...
[... 193 lines cut ...]
==23521==    by 0x604DFA: Rsyn::Session::evaluateFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x615C74: Rsyn::Shell::runScript(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x4B76E6: main (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521== 
==23521== Conditional jump or move depends on uninitialised value(s)
==23521==    at 0x8C4A8D: FastRoute::routeLVEnew(int, int, int, int) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x8C60BA: FastRoute::routeLVAll(int, int) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x88110D: FastRoute::FT::run(std::vector<FastRoute::NET, std::allocator<FastRoute::NET> >&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x628088: Rsyn::FastRouteProcess::run(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator> const&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x602278: Rsyn::Session::runProcess(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator> const&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x604068: std::_Function_handler<void (ScriptParsing::Command const&), Rsyn::Session::registerDefaultCommands()::{lambda(ScriptParsing::Command const&)#4}>::_M_invoke(std::_Any_data const&, ScriptParsing::Command const&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x60FC56: ScriptParsing::CommandManager::evaluateCommand(ScriptParsing::ParsedCommand const&) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
[... cut ...]
==23521==    by 0x615C74: Rsyn::Shell::runScript(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521==    by 0x4B76E6: main (in /home/oyvind/FastRoute4-lefdef/FRlefdef)
==23521== 
==23521== 
==23521== More than 10000000 total errors detected.  I'm not reporting any more.
==23521== Final error counts will be inaccurate.  Go fix your program!
==23521== Rerun with --error-limit=no to disable this cutoff.  Note
==23521== that errors may occur in your program without prior warning from
==23521== Valgrind, because errors are no longer being displayed.
==23521== 

```

oharboe, 2019-08-06:
@tajayi has the input files to this test. The files are confidential, but hopefully @tajayi  can provide some details so that the problem can be tracked down.

eder-matheus, 2019-08-06:
Hi, @oharboe.

Can you compile FastRoute4-lefdef in debug mode and send to us the gdb backtrace?

To compile FastRoute4-lefdef in debug mode, just add 'DCMAKE_BUILD_TYPE=Debug' to the optional parameters for cmake in the make command.

oharboe, 2019-08-06:
Using:  
[Dockerfile.zip](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/files/3473983/Dockerfile.zip)

```
$ gdb --args /tmp/FRlefdef --no-gui --script `pwd`/fastroute.rsyn
initial grid wrong y1 x1 [0 0] , net start [10 0] routelen 305
 checking failed 0

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x44000001e14) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x44000001e14) at malloc.c:3103
#1  0x0000555555cf587b in FastRoute::reInitTree (netID=0) at /FastRoute4-lefdef/third_party/fastroute/src/maze.cpp:963
#2  0x0000555555cfa598 in FastRoute::mazeRouteMSMD (iter=1, expand=9, costHeight=34, ripup_threshold=-1, mazeedge_Threshold=0, Ordering=0 '\000', 
    cost_type=1) at /FastRoute4-lefdef/third_party/fastroute/src/maze.cpp:1643
#3  0x0000555555ccd6dd in FastRoute::FT::run (this=0x5555697cd328, result=std::vector of length 0, capacity 0)
    at /FastRoute4-lefdef/third_party/fastroute/src/FastRoute.cpp:649
#4  0x0000555555a74089 in Rsyn::FastRouteProcess::run (this=<optimized out>, params=...)
    at /FastRoute4-lefdef/third_party/rsyn/src/tool/fastroute/FastRoute.cpp:113
#5  0x0000555555a4e279 in Rsyn::Session::runProcess (name="rsyn.fastRoute", params=...)
    at /FastRoute4-lefdef/third_party/rsyn/src/session/Session.cpp:514
#6  0x0000555555a50069 in Rsyn::Session::<lambda(const ScriptParsing::Command&)>::operator() (__closure=<optimized out>, command=...)
    at /FastRoute4-lefdef/third_party/rsyn/src/session/Session.cpp:216
#7  std::_Function_handler<void(const ScriptParsing::Command&), Rsyn::Session::registerDefaultCommands()::<lambda(const ScriptParsing::Command&)> >::_M_invoke(const std::_Any_data &, const ScriptParsing::Command &) (__functor=..., __args#0=...) at /usr/include/c++/7/bits/std_function.h:316
#8  0x0000555555a5bc57 in std::function<void (ScriptParsing::Command const&)>::operator()(ScriptParsing::Command const&) const (__args#0=..., 
    this=0x7fffffffd8f0) at /usr/include/c++/7/bits/std_function.h:706
#9  ScriptParsing::CommandManager::evaluateCommand (this=0x5555695bef20, parsedCommand=...)
    at /FastRoute4-lefdef/third_party/rsyn/./src/script/ScriptCommand.h:1071
#10 0x0000555555a50dfb in ScriptParsing::CommandManager::evaluateFile (filename=..., this=0x5555695bef20)
    at /FastRoute4-lefdef/third_party/rsyn/./src/script/ScriptCommand.h:1227
#11 Rsyn::Session::evaluateFile (filename=...) at /FastRoute4-lefdef/third_party/rsyn/src/session/Session.cpp:155
#12 0x0000555555a61c75 in Rsyn::Shell::runScript (this=<optimized out>, startupScript=..., interactive=<optimized out>)
    at /FastRoute4-lefdef/third_party/rsyn/src/shell/Shell.cpp:30
#13 0x00005555559036e7 in main (argc=<optimized out>, argv=0x7fffffffde48) at /FastRoute4-lefdef/third_party/rsyn/src/main.cpp:251

(gdb) up
#1  0x0000555555cf587b in FastRoute::reInitTree (netID=0) at /FastRoute4-lefdef/third_party/fastroute/src/maze.cpp:963
963	/FastRoute4-lefdef/third_party/fastroute/src/maze.cpp: No such file or directory.
(gdb) info locals
deg = 2
numEdges = 1
edgeID = 0
d = 0
j = 0
treeedge = 0x55556d74cf90
rsmt = {deg = 0, length = 0, branch = 0x0}
x = {0 <repeats 144 times>, 3, 4, 2, 6, 5, 1, 0, 0, 5, 5, 6, 7, 3, 1, 14, 15, 17, 13, 0, 4, 18, 12, 30, 2, 24, 19, 20, 17, 15, 16, 28, 31, 26, 22, 
  9, 12, 5, 5, 8, 9, 46, 49, 59, 20, 44, 51, 15, 19, 3, 57, 4, 42, 5, 11, 6, 50, 53, 52, 54, 43, 53, 52, 54, 43, 0 <repeats 936 times>, 
  1171089186, 1171089186, 1169536968, 1159074371, 1159596143, 1163222350, 1167984751, 1162178806, 1169276082, 1167462979, 1167136349, 1125926893, 
  1127081574, 1124269924, 1134315501, 1125888331, 1121457275, 1127068416, 1125471035, 1125053738, 1134449685, 1125399230, 1124981932, 1132643655, 
  1134205134, 1136356656, 1136738050, 1133579188, 1136320752, 1136702146, 1135313413, 1138054977, 1131847024, 1135178831, 1120999808, 1136066821, 
  1131811204, 1130616254, 1135107572, 1134926685, 1131197284, 1134981499, 1138133316, 1138369017, 1128917012, 1135507715, 1134493652, 1139076120, 
  1127891573, 1143441068, 1142846042, 1141385649, 1141875552, 1145284904, 1145133061, 1144189840, 1148390571, 1144037997, 1134152731, 1133907308, 
  1132623695, 1131463912, 1135870333, 1129236392, 1126253892, 1128254701, 1127763856, 1126027183, 1129689095, 1134567648, 1132029610, 1135322630, 
  1135284845, 1137738716, 1130896779, 1129160106, 1134095515, 1134265368, 1132811902, 1136313206, 1141588251, 1141742402, 1139972106, 
  0 <repeats 241 times>, -61170, 32767, -61184, 32767, -62208, 32767, 1440404000, 21845, 1814036352, 21845, 1814036384, 21845, -62224, 32767, 
  1439500332, 21845, 0, 0, 1814036384, 21845, -62192, 32767, 1439495734...}
y = {0 <repeats 144 times>, -963679280, -804905914, -963646596, -804905914, -962661465, -969630999, -962723568, -968099831, -962106135, 
  -961531896, -973903475, -960971475, -964442411, -974425950, -995771343, -997635842, -993375847, -999701864, -996932484, -1002164929, -994957727, 
  -1000412085, -999769289, -999464274, -997748263, -997326161, -995223495, -996918354, -996330459, -998801744, -996517789, -998296944, -995412039, 
  -997639198, -995558245, -999859116, -995267354, -804905914, -1001876554, -1003419615, -998416486, -1001531122, -998953126, -1002660944, 
  -998003297, -1005378490, -1003686022, -1005708315, -998146663, -1018835744, -996537258, -1016510176, -996747830, -994908717, -990407654, 
[... cut ...]
  1, -45168, 32767, 0, 0, -45648, 32767, 0, 0, 8, 2147483647, 15...}
(gdb) list
958	in /FastRoute4-lefdef/third_party/fastroute/src/maze.cpp
(gdb) print numEdges
$1 = 1
(gdb) print edgeID
$2 = 0
(gdb) 
$3 = 0
(gdb) print treeedge
$4 = (FastRoute::TreeEdge *) 0x55556d74cf90
(gdb) print *treeedge
$5 = {assigned = 1 '\001', len = 229, n1 = 0, n1a = 0, n2 = 1, n2a = 1, route = {type = 3 '\003', xFirst = 0 '\000', HVH = 0 '\000', Zpoint = 1, 
    gridsX = 0x55556ed0d0d0, gridsY = 0x55556ed0d340, gridsL = 0x44000001e14, routelen = 305}}
(gdb) 

```

oharboe, 2019-08-06:
It looks like gridsL contains junk:  0x44000001e14

eder-matheus, 2019-08-09:
Is it possible for you to create a minimal testcase that reproduces this crash and shares it with us?

tajayi, 2019-08-10:
@oharboe Can this be closed as a duplicate of #8 ?

Thanks

closed 2019-08-10 by eder-matheus