The-OpenROAD-Project-Attic/FastRoute #7: lefdef print message when you have illegal def (specifically multidriven pins)

tajayi, 2019-08-10:
**Priority:** Medium

**Details:**
fastroute should be able to detect illegal defs, specifically ones that have multiple nets connected to the same sink/pin. TritonCTS had/has a bug that can create this, and the resulting case is TritonRoute complaining about no coverage on one of the nets (I think fastroute only covers the pin on one net)

**Steps to reproduce**
Try to route a design with a pin connected to multiple nets. Ideally this should be added as a unit test that is expected to fail

oharboe, 2019-08-11:
tiny-tests need support for annotations in comments about the expected error message in the case a test is expected to fail.

Look at a couple of these tests from gcc-testsuite, it's a folder with thousands of tiny tests for the gcc compiler. It's absolutely fanatastic to work with as a compile developer.

One of the things that you can do that's very useful is to specify options in a .c file and specify what expected error message you should see.

https://github.com/gcc-mirror/gcc/tree/master/gcc/testsuite/gcc.c-torture/execute

eder-matheus, 2019-08-15:
Please, check commit 237ee4bff34060fd7e1e322086c259ee5c51a548.

tajayi, 2019-09-09:
Closing based on commit information. The bug has been fixed in TritonCTS and is hard to manually create a bad test case.

closed 2019-09-09 by tajayi