The-OpenROAD-Project-Attic/FastRoute #9: fastroute should gracefully die on illegal macro placement

tajayi, 2019-08-10:
**Priority:** Medium

**Details:**
There is currently a bug in fplan that can result in macros being placed outside of the design area or on top of each other. This results in a mysterious crash by fastroute. The tool should be able to detect this bad behavior and gracefully die

oharboe, 2019-08-11:
I would change "gracefully die" to be more specific: "produce a helpful error message"

isadorasop, 2019-08-13:
Please, check commit 4aa46b16e85b5a628497ac1e6d3784a4f786c6cf.

tajayi, 2019-08-15:
Validated

```

----------------
Checking pin placement...
Checking pin placement... Done!
Checking macros...
ERROR: There are macros outside the die area.
ERROR: There are macros outside the die area.
ERROR: There are macros outside the die area.
ERROR: There are macros outside the die area.
ERROR: There are macros outside the die area.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
ERROR: There are macros that overlap eachother.
make: *** [results/tsmc65lp/bp_fe_top/route.guide] Error 254
make: Leaving directory `/net/trenton/x/ajayi/projects/OpenROAD/alpha-release/flow'

tajayi, 2019-08-15:
Ooops... Looks like it fails to check "valid" floorplans
[image]

```
Checking pin placement...
Checking pin placement... Done!
Checking macros...
ERROR: There are macros outside the die area.
ERROR: There are macros that overlap eachother.
make: *** [results/tsmc65lp/bsg_manycore_tile/route.guide] Error 254
make: Leaving directory `/net/trenton/x/ajayi/projects/OpenROAD/alpha-release/flow'
```

My suspicion is that the math may not comprehend rotations. All of the macros in the design above are rotated R90.

PS, if it's easy to print the exact macro name that's violating, that might be helpful

oharboe, 2019-08-15:
could the makefile flow write out this picture to results/?

tajayi, 2019-08-15:
The image is generated using a commercial tool/software. Long term, I plan to integrate klayout to generate similar pictures and will probably put them in the results folder.

oharboe, 2019-08-15:
gotit.

a bit of bling in reporting goes a long way. i would urge reporting to be
compatible w continuous integration: static files with no gui. perhaps a
top level html file?

vvbandeira, 2019-08-18:
@tajayi, can you test with the new version on the master branch (dd29747029fe0e4837da8696153998d5dd91aa89)?

Thanks for your feedback.

tajayi, 2020-01-26:
Closing this for now

eder-matheus, 2020-04-01:
FastRoute generates warning messages for any object out of the die area, pointing the name of the instances with this issue. No segfault with the available testcases

closed 2020-04-01 by eder-matheus