The-OpenROAD-Project-Attic/FastRoute #10: Ask for help

wjx-1999, 2019-08-23:
Hello, I have some questions to ask your help

1. I used FastRoute4 to test my own placement, only part of which could be successfully run, and the rest would have segment fault(段错误) after several iterations
[image]

2.There seems to be a problem with the generated guide files.When these guide files are put into the detailed router, warning message FAIL_DETACHED_GUIDE appears,because of disconnected guides for those nets, i.e., the pins cannot be connected by on-track routing only.

Do you need me to provide my placement input(LEF/DEF) and detailed router to debug your code?

I want to know how to solve these problems.Thanks！

wjx-1999, 2019-08-24:
> Hi, @wjx-1999. It would be great if you can provide us your LEF/DEF input files. It will make a way easier for us to debug the problem.
>
> Just a question: in 1) you show a segfault problem, but in 2) you're talking about generated guides file. These two situations are with different testcases, right?
>
>
> 从QQ邮箱发来的超大附件
>
> benchmark.zip (77.38M, 2019年09月23日 09:59 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=7364353708ba4fcda903ba6b1561541904560455045004031c500504524c5252095418560a535e1b095256030b57560755565104336c6654540a565f5e00145d1f1e5c47335c&t=exs_ftn_download&code=1d573af6

wjx-1999, 2019-08-24:
Hello,@eder-matheus .The two situations are with two different testcases.

Some tests can be run successfully but the guide file still has defects, and the rest will have segment fault that cannot be run successfully.I hope you can solve the problem of segment fault. 

I'll provide you with the input file(benchmark.zip). Thank you!

wjx-1999, 2019-08-24:
[http://mail.qq.com/cgi-bin/ftnExs_download?k=7364353708ba4fcda903ba6b1561541904560455045004031c500504524c5252095418560a535e1b095256030b57560755565104336c6654540a565f5e00145d1f1e5c47335c&t=exs_ftn_download&code=1d573af6](url)
You can download the input file from the website.If not, please inform me again. @eder-matheus

wjx-1999, 2019-08-24:
http://mail.qq.com/cgi-bin/ftnExs_download?k=7364353708ba4fcda903ba6b1561541904560455045004031c500504524c5252095418560a535e1b095256030b57560755565104336c6654540a565f5e00145d1f1e5c47335c&t=exs_ftn_download&code=1d573af6

You can download the input file from the website.If not,inform me again. @eder-matheus

wjx-1999, 2019-08-29:
Hello,@eder-matheus 
May I ask if you have solved the problem of segment fault?

vvbandeira, 2019-08-29:
Hi @wjx-1999, we are working on a fix. As soon as the problem is fixed and tested we will reply here pointing to a commit for you to test. 

Thanks,
-- 
Vitor Bandeira

eder-matheus, 2020-04-01:
FastRoute was updated to avoid segfaults when handle overflow. Please, check the [enhancements](https://github.com/The-OpenROAD-Project/FastRoute/tree/enhancements) branch of the repository

Note: this branch needs to be used inside the OpenROAD app. More information [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/README.md)

closed 2020-04-01 by eder-matheus