The-OpenROAD-Project-Attic/FastRoute #12: Minimum routing layer

ax3ghazy, 2019-11-06:
Is it possible to limit the minimum routing layer available for FR to use? (e.g., need to limit the amount of routing that happens in the lowest layer, so that the detailed router only uses vias from upper layers to reach that layer).
Thank you.

eder-matheus, 2019-11-06:
Hi, @ax3ghazy. We have this feature in develop branch. 
Note that we have some differences in the script to run FastRoute4-lefdef.
You can find sample scripts [here](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/tree/develop/support/scripts).

ax3ghazy, 2019-11-08:
Hi @eder-matheus ,
Thank you!
Another related question: is it possible to somehow force the capacity for a certain layer to be low to force the router to use it as scarcely as possible? I have tried the adjustment parameter, but I cannot seem to get reliable results. Thanks again.

closed 2019-11-08 by ax3ghazy