The-OpenROAD-Project-Attic/FastRoute #13: Adjusting capacity per layer

ax3ghazy, 2019-11-20:
Hello,

Currently only a single capacity adjustment parameter can be specified. It would be useful to be able to simply reduce the capacity of a specific layer in the .rsyn config (or the tcl interface in develop). I am aware of "set_region_adjustment" in the develop branch, but I think a separate "set_layer_adjustment", where you don't have to specify the region, would be very useful.

Thank you.

eder-matheus, 2019-11-20:
Hi, @ax3ghazy.
I added this feature in branch [opendb-integration](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/tree/opendb-integration). You can find a script example [here](https://github.com/The-OpenROAD-Project/FastRoute4-lefdef/blob/opendb-integration/etc/scripts/example.tcl).

Notice that this branch doesn't have Rsyn anymore. So only TCL scripts can be used.

We will be merging this branch with develop and master branches soon, so I recommend to use this version now.

Thanks.

ax3ghazy, 2019-11-20:
Great work, thanks. I will be testing it soon:)

closed 2019-11-20 by ax3ghazy

eder-matheus, 2019-11-22:
Hi, @ax3ghazy.

Just for your information, I already merged "opendb-integration" branch to "develop" and "master" branches. You can use these two branches for now.

Thanks.

ax3ghazy, 2019-11-23:
Hi, @eder-matheus,

That's great; thanks for letting me know!