The-OpenROAD-Project-Attic/FastRoute #16: fastroute won't run if fewer than 3 metal layers

teadotjay, 2020-01-14:
My technology has only two routing layers. To get fastroute to run, I had to create a third dummy metal layer and the associated tracks, then set -max_routing_layer 2.

```tcl
% fastroute -max_routing_layer 2
WARNING: Default output guide name: out.guide
WARNING: Default capacity adjustment: 0.0 (no adjustment)
WARNING: Default minimum layer: 1
Initializing grid...
[ERROR] Layer3 not found! Exiting...
```

eder-matheus, 2020-01-14:
Hi, @teadotjay.

Please, check the commit d836a4a160b6feeb46d5c9db40cbc5d887f94582 of the OpenROAD app. I believe it should solves your issue.

teadotjay, 2020-02-04:
I tried building this version, but there are some new dependencies that are not handled by the Dockerfile, and I couldn't find any instructions for installing them on CentOS 7.

rovinski, 2020-02-04:
Those are all optional dependencies; they should not be required to successfully build the tools and run through the flow.

eder-matheus, 2020-04-01:
I'm closing this issue since it was tested. 
@teadotjay, please let me know if you still have issues regarding metal layers

closed 2020-04-01 by eder-matheus