The-OpenROAD-Project-Attic/FastRoute #19: OpenROAD is crashing while starting FastRoute

tamimcse, 2020-03-24:
OpenROAD is crashing while starting FastRoute for https://github.com/tamimcse/test/blob/master/top.v. The Verilog is generated using Bambu HLS. The OpenROAD script is https://github.com/tamimcse/test/blob/master/bash_script.sh. The output is as following: You will see the crash at the very end.

`.....
......
yosys> opt_clean -purge

Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy..
Removed 96 unused cells and 9494 unused wires.
<suppressed ~183 debug messages>
y_typical.lib iberty /opt/panda/share/panda//nangate45/lib/NangateOpenCellLibrary

Printing statistics.
=== _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy ===

Number of wires: 20066
Number of wire bits: 21196
Number of public wires: 2469
Number of public wire bits: 3599
Number of memories: 0
Number of memory bits: 0
Number of processes: 0
Number of cells: 18580
AND2_X1 1925
AND2_X2 49
AND2_X4 45
AND3_X1 488
AND3_X2 3
AND3_X4 6
AND4_X1 120
AND4_X2 3
AND4_X4 3
AOI211_X1 272
AOI211_X2 3
AOI21_X1 882
AOI21_X2 3
AOI221_X1 1
AOI221_X2 2
AOI221_X4 33
AOI22_X1 22
BUF_X1 4070
BUF_X16 2
BUF_X2 115
BUF_X4 273
BUF_X8 5
CLKBUF_X2 58
CLKBUF_X3 3
DFF_X1 1514
INV_X1 807
INV_X2 18
INV_X32 5
INV_X4 6
LOGIC0_X1 1
MUX2_X1 1143
MUX2_X2 4
NAND2_X1 1288
NAND2_X2 3
NAND2_X4 1
NAND3_X1 451
NAND3_X2 1
NAND4_X1 84
NAND4_X2 1
NOR2_X1 1262
NOR2_X2 11
NOR2_X4 9
NOR3_X1 335
NOR3_X2 1
NOR3_X4 1
NOR4_X1 58
NOR4_X2 2
[... 445 lines cut ...]
fixed cells : 1716
nets : 22618
design area : 1438127936000.000
total fixed area : 1825824000.000
total movable area : 287720496000.000
design utilization : 20.032
rows : 428
row height : 2800
Check Legality
row check ==> PASS
site check ==> PASS
power check ==> PASS
edge_check ==> PASS
placed_check ==>> PASS
overlap_check ==> PASS
-------------------- Placement Analysis ------------------------
total displacement : 1657681
average displacement : 76
max displacement : 16853
original HPWL : 763228.133
legalized HPWL : 763580.620
delta HPWL : 0%
Adjust layer 2 in 70.0%
Adjust layer 3 in 70.0%
*** buffer overflow detected ***: openroad terminated`

eder-matheus, 2020-04-01:
Hi, @tamimcse.

Are you generating the output files from each step (e.g: 3_placed.def after placement, 4_cts.def after clock tree synthesis, etc.)? If so, can you send me the input files for FastRoute?

tamimcse, 2020-04-01:
I added the DEF files at https://github.com/tamimcse/test/tree/master/OpenROAD_results. I also updated the openroad script (https://github.com/tamimcse/test/blob/master/bash_script.sh). Please let me know if you need any other information.

eder-matheus, 2020-04-02:
When I try to run "bash_script.sh", I get these following errors:
./bash_script.sh: 14: ./bash_script.sh: /scripts/mergeLef.py: not found
./bash_script.sh: 15: ./bash_script.sh: /scripts/padLefMacro.py: not found
./bash_script.sh: 16: ./bash_script.sh: /scripts/modifyLefSpacing.py: not found
./bash_script.sh: 17: ./bash_script.sh: /scripts/mergeLib.pl: not found
./bash_script.sh: 18: ./bash_script.sh: /scripts/markDontUse.py: not found

If you can send me these files, and if you can send me the LEF files used as input for FastRoute, it would be great

rovinski, 2020-04-02:
@tamimcse if you use OpenROAD-flow, you can run `make global_route_issue` which will produce a tar.gz file with everything @eder-matheus needs to debug. Please browse the tar to make sure it doesn't package any proprietary / NDA materials or things you do not want to share. If you are able to reproduce the issue with nangate45 on an open-source netlist, then there should be no IP issues.

tamimcse, 2020-04-02:
OK, never mind about the previous scripts. I added all the scripts in https://github.com/tamimcse/test/tree/master/sail_sim. Here  you want to run synthesize_Synthesis__Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy.sh. It sets some environment variables, then runs HLS_output/Synthesis/bash_flow/bash_script.sh. This is the actual openroad script. It generates output in OpenROAD_results folder. Here you will find the intermediate DEF files.

eder-matheus, 2020-04-02:
I'm having some trouble running your script. Does it need to be run in a Docker or something like that? Also, do I need to clone and build the OpenROAD-flow at some place inside your repo?

tamimcse, 2020-04-02:
No, you don't need OpenROAD-flow at all. However you will need to install openroad, yosys and TritonRoute. Those binaries needs to be accessible via PATH variable (they are accessible by default). 

Then you simply can clone `https://github.com/tamimcse/test`. Then enter `test/sail_sim/` and then run `sh synthesize_Synthesis__Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy.sh`. This should run openroad. 

Here we don't use the TCL scripts of OpenROAD-flow. We rather use shell script (`HLS_output/Synthesis/bash_flow/bash_script.sh`) which is analogous to  OpenROAD-flow TCL scripts.

eder-matheus, 2020-04-16:
Hi, @tamimcse.

Sorry for the delay with it. I spent the last days fixing some memory issues in FastRoute code, that was probably the cause of the crash that you are facing. Can you try to update your submodule of FastRoute with the last "openroad" branch updates?

tamimcse, 2020-04-20:
Thanks @eder-matheus for looking into the issue!!!!

I updated OpenROAD to "openroad" branch along with all the sub-modules as following:
`git fetch origin
git checkout -f --recurse-submodules  origin/openroad`. 

The new version is successfully compiled and installed . But I'm still getting the same error. But the output is bit different. The output is as following:
`  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
   0.00    0.00   clock network delay (ideal)
   0.00    0.00 ^ _34076_/CK (DFF_X1)
   0.08    0.08 ^ _34076_/Q (DFF_X1)
   0.21    0.29 ^ _31306_/Z (BUF_X1)
   0.05    0.35 v _20241_/ZN (INV_X1)
   0.07    0.42 v _20242_/Z (MUX2_X1)
   0.04    0.46 ^ _20243_/ZN (NOR2_X1)
   0.04    0.50 v _20244_/ZN (NAND3_X1)
   0.05    0.55 ^ _20246_/ZN (NOR2_X1)
   0.03    0.57 v _30089_/ZN (XNOR2_X1)
   0.03    0.60 v _30092_/ZN (AND2_X1)
   0.04    0.64 ^ _30094_/ZN (NOR2_X1)
   0.01    0.66 v _30097_/ZN (NOR2_X1)
   0.07    0.73 v _30103_/ZN (OR3_X1)
   0.02    0.74 ^ _30109_/ZN (NAND3_X1)
   0.04    0.78 ^ _30111_/ZN (AND2_X1)
   0.01    0.79 v _30114_/ZN (INV_X1)
   0.03    0.82 v _30115_/ZN (AND2_X1)
   0.04    0.85 ^ _30117_/ZN (NOR2_X1)
   0.01    0.86 v _30120_/ZN (INV_X1)
   0.03    0.89 v _30121_/ZN (AND2_X1)
   0.04    0.93 ^ _30123_/ZN (NOR2_X1)
   0.01    0.94 v _30126_/ZN (INV_X1)
   0.03    0.97 v _30127_/ZN (AND2_X1)
   0.03    1.00 ^ _30138_/ZN (OAI211_X1)
   0.04    1.04 ^ _30141_/ZN (AND2_X1)
   0.01    1.05 v _30146_/ZN (NOR3_X1)
   0.04    1.09 ^ _30147_/ZN (NOR2_X1)
   0.01    1.09 v _30151_/ZN (INV_X1)
   0.03    1.13 v _30152_/ZN (AND2_X1)
   0.04    1.16 ^ _30154_/ZN (NOR2_X1)
   0.01    1.18 v _30162_/ZN (INV_X1)
   0.02    1.19 ^ _30178_/ZN (NAND4_X1)
   0.05    1.24 ^ _30182_/ZN (AND2_X1)
   0.01    1.25 v _30191_/ZN (INV_X1)
   0.03    1.28 v _30208_/ZN (AND4_X1)
   0.04    1.32 ^ _30213_/ZN (NOR2_X1)
   0.01    1.33 v _30216_/ZN (INV_X1)
   0.04    1.37 v _30240_/ZN (AND4_X1)
   0.03    1.41 ^ _30257_/ZN (OAI211_X1)
   0.05    1.45 ^ _30259_/ZN (AND2_X1)
   0.01    1.46 v _30269_/ZN (INV_X1)
   0.03    1.50 v _30270_/ZN (AND3_X1)
   0.03    1.53 ^ _30276_/ZN (OAI21_X1)
   0.02    1.55 v _30279_/ZN (NAND2_X1)
   0.04    1.59 v _30281_/ZN (XNOR2_X1)
   0.03    1.61 v _33350_/Z (BUF_X1)
   0.00    1.61 v _34112_/D (DFF_X1)
           1.61   data arrival time

  10.00   10.00   clock clock (rise edge)
   0.00   10.00   clock network delay (ideal)
   0.00   10.00   clock reconvergence pessimism
          10.00 ^ _34112_/CK (DFF_X1)
  -0.04    9.96   library setup time
           9.96   data required time
---------------------------------------------------------
           9.96   data required time
          -1.61   data arrival time
---------------------------------------------------------
[... 198 lines cut ...]
    Num wire segments: 4994
    Num keys in characterization LUT: 1677
    Actual min input cap: 8
 Reading solution list file "/opt/panda/share/panda//nangate45/tritonCTS/sol_list.txt"
 **********************
 *  Find clock roots  *
 **********************
 User did not specify clock roots.
 Using OpenSTA to find clock roots.
 Looking for clock sources...
    Clock names: clock 
 ************************
 *  Populate TritonCTS  *
 ************************
 Initializing clock nets
 Number of user-input clocks: 1 ( "clock" )
 Looking for clock nets in the design
 Net "clock" found
clock
invalid command name "legalize_placement"
Adjust layer 2 in 70.0%
Adjust layer 3 in 70.0%
*** buffer overflow detected ***: openroad terminated
`
Note that if I change `uint32_t C112[SIZE112]` to `uint8_t C112[SIZE112]` in https://github.com/tamimcse/test/blob/master/module.cpp, the generated verilog works properly with OpenROAD. So I guess, it may causing error at some boundary condition or exceeding memory..

rovinski, 2020-04-20:
Hi @tamimcse, you can use three tick marks ( \`\`\` ) before and after text to create a multi-line code block. It helps with your log readability, thanks.

\`\`\`
example
example
example
\`\`\`

```
example
example
example
```

tamimcse, 2020-05-14:
This error doesn't occur if compiler optimization is turned off. I compiled openroad using following command and it worked fine.
```
cmake -DCMAKE_BUILD_TYPE=DEBUG \
      -DCMAKE_C_FLAGS_DEBUG="-g -O0" \
      -DCMAKE_CXX_FLAGS_DEBUG="-g -O0"
```
Probably the error is occurring not due to exceeding memory but for other causes. 

Few days ago, I encountered  `buffer overflow detected` while working on an another project. It was working fine in the compiler unoptimized version, but when turn on compiler optimization, I got `buffer overflow detected`. The cause was as following:

```
int insert_something(int a) {
  //return statement is missing here.
  insert_something_inner (a);
}

int insert_something_inner(int a) {
  //body of the function
}
```

After adding the `return` statement, it `buffer overflow detected` error disappeared. I guess, FastRoute may have similar issue.

eder-matheus, 2020-05-14:
I fixed a lot of memory issues in FastRoute. Are you facing this issue with the latest version of the repository?

tamimcse, 2020-05-14:
Yeah, this issue still exists (if I use compiler optimization). Here is the output of `openroad` branch:

OpenROAD-Flow makefile, sdc and verilog would be found at https://github.com/tamimcse/test5

```
tamim@tamim-HP-ENVY-17-Notebook-PC:~/OpenROAD-flow/flow$ make
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" openroad -no_init -exit ./scripts/global_place.tcl) 2>&1 | tee ./logs/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/3_1_place_gp.log
OpenROAD 1.1.0 ef1118f201
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0: 
Reading DEF file: ./results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/2_floorplan.def
Notice 0: Design: _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy
Notice 0:     Created 1173 pins.
Notice 0:     Created 21392 components and 100636 component-terminals.
Notice 0:     Created 5 special nets and 42784 connections.
Notice 0:     Created 21032 nets and 57852 connections.
Notice 0: Finished DEF file: ./results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/2_floorplan.def
[INFO] DBU = 2000
[INFO] SiteSize = (380, 2800)
[INFO] CoreAreaLxLy = (20140, 22400)
[INFO] CoreAreaUxUy = (1680360, 1680000)
[INFO] NumInstances = 21392
[INFO] NumPlaceInstances = 18426
[INFO] NumFixedInstances = 2966
[INFO] NumDummyInstances = 0
[INFO] NumNets = 21032
[INFO] NumPins = 59025
[INFO] DieAreaLxLy = (0, 0)
[INFO] DieAreaUxUy = (1700260, 1701600)
[INFO] CoreAreaLxLy = (20140, 22400)
[INFO] CoreAreaUxUy = (1680360, 1680000)
[INFO] CoreArea = 2751980672000
[INFO] NonPlaceInstsArea = 3155824000
[INFO] PlaceInstsArea = 184237984000
[INFO] Util(%) = 6.702427
[INFO] StdInstsArea = 184237984000
[INFO] MacroInstsArea = 0
[InitialPlace]  Iter: 1 CG Error: 9.40766e-05 HPWL: 2055551700
[InitialPlace]  Iter: 2 CG Error: 6.13227e-05 HPWL: 1573406637
[InitialPlace]  Iter: 3 CG Error: 4.61366e-05 HPWL: 1558483869
[InitialPlace]  Iter: 4 CG Error: 3.50621e-05 HPWL: 1539441037
[InitialPlace]  Iter: 5 CG Error: 2.49481e-06 HPWL: 1533589517
[INFO] FillerInit: NumGCells = 85894
[INFO] FillerInit: NumGNets = 21032
[INFO] FillerInit: NumGPins = 59025
[INFO] TargetDensity = 0.300000
[INFO] AveragePlaceInstArea = 9998805
[INFO] IdealBinArea = 33329348
[INFO] IdealBinCnt = 82569
[INFO] TotalBinArea = 2751980672000
[INFO] BinCnt = (256, 256)
[INFO] BinSize = (6486, 6475)
[INFO] NumBins = 65536
[NesterovSolve] Iter: 1 overflow: 0.963505 HPWL: 1419871020
[NesterovSolve] Iter: 10 overflow: 0.816161 HPWL: 1814183015
[NesterovSolve] Iter: 20 overflow: 0.803429 HPWL: 1766025678
[NesterovSolve] Iter: 30 overflow: 0.797274 HPWL: 1769386195
[NesterovSolve] Iter: 40 overflow: 0.795752 HPWL: 1763412967
[NesterovSolve] Iter: 50 overflow: 0.795315 HPWL: 1762737228
[... 522 lines cut ...]
This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef
Notice 0: Reading LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  ./platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef
Notice 0: 
Reading DEF file: ./results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/4_cts.def
Notice 0: Design: _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy
Notice 0: 		Created 100000 Insts
Notice 0:     Created 1173 pins.
Notice 0:     Created 155487 components and 371596 component-terminals.
Notice 0:     Created 5 special nets and 310974 connections.
Notice 0:     Created 22417 nets and 60622 connections.
Notice 0: Finished DEF file: ./results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/4_cts.def
*** buffer overflow detected ***: openroad terminated
Command terminated by signal 6
0:06.82elapsed 98%CPU 387316memKB
Makefile:326: recipe for target 'results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/route.guide' failed
make: *** [results/nangate45/_Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy/route.guide] Error 134
```

eder-matheus, 2020-05-14:
I see. Sorry to ask for it again, but can you point me how do I reproduce this issue on my side, step by step? (e.g., what repositories I should clone, how should I the directories, etc...)

tamimcse, 2020-05-14:
This is how I did it:

1. Installed OpenROAD `openroad` branch
2. Also installed Yosys and TritonRoute from source code. Installing Yosys via `apt-get` doesn't work.
2. Cloned OpenROAD-Flow `openroad` branch
3. I went to `OpenROAD-flow/flow/designs/src`. Created a folder same `sail`. Entered inside `sail`. Cloned https://github.com/tamimcse/test5 there. This will fetch the makefile, sdc and verilog file.
4. Then opened `OpenROAD-flow/flow/Makefile`. Set `DESIGN_CONFIG ?= ./designs/src/sail/sail.mk`. It points to `gcd.mk` by default, so we needed to update it. 
5. Now execute `source setup_env.sh` in `OpenROAD-Flow`
6. Then execute `make` in `OpenROAD-Flow/flow`
The `make` will execute OpenROAD flow on our files.

eder-matheus, 2020-05-14:
Thank you very much! I will try it on my end and let you know when I have updates