The-OpenROAD-Project-Attic/FastRoute #20: Macros of class PAD_* don't have their signals routed

rbarzic, 2020-04-01:
While trying to route a design  with pre-placed I/O cells (also called sometimes "PADs")  with OpenROAD, I noticed that the signals to/from those cells were not routed. I tracked down the problem to the fact that those signals were not included in the route.guide file generated by the fastroute command.
The offending code seems to be the following:  (src/DBWrapper.cpp around line 359)

```
                       if (master->getType() == odb::dbMasterType::PAD ||
                            master->getType() == odb::dbMasterType::PAD_INPUT ||
                            master->getType() == odb::dbMasterType::PAD_OUTPUT ||
                            master->getType() == odb::dbMasterType::PAD_INOUT ||
                            master->getType() == odb::dbMasterType::PAD_POWER ||
                            master->getType() == odb::dbMasterType::PAD_SPACER) {
                                padFound = true;
                                break;
                        }`
```
Macros of class PAD, PAD_INPUT, ... seem to have their pins excluded from the route.guide file
If I remove the first four compare statements, I get a correct routing of the signals going to/from the I/O cells.

eder-matheus, 2020-04-01:
Hi, @rbarzic.

Can you check commit 31954dfb907edfe842764711cea079c3c05bb709? I've added an option to allow routing for nets with PADs. Just add the flag `-route_nets_with_pad ` to enable it.

Thanks

eder-matheus, 2020-04-01:
Also, look for these changes in [enhancements](https://github.com/The-OpenROAD-Project/FastRoute/tree/enhancements) branch

rbarzic, 2020-04-08:
Hi Eder
Sorry for my late response but building the lastest version of OpenROAD has been difficult for me the last days - Finally I managed to get it to build  using the openroad branch of FastRoute and I can confirm I have the correct routing of signals to the I/O cells
Thank you very much

closed 2020-04-08 by rbarzic