The-OpenROAD-Project-Attic/FastRoute #27: Guides vector is empty!!!

tamimcse, 2020-05-14:
Hi

I'm getting error `Guides vector is empty!!!` error in FastRoute in OpenROAD for https://github.com/tamimcse/test/blob/master/sail_sim/top.v. The output is as following:

```
yosys> exit

Warnings: 8 unique messages, 72 total
End of script. Logfile hash: 265393fcec, CPU: user 25.91s system 0.11s, MEM: 105.03 MB peak
Yosys 0.9+1706 (git sha1 b7419544, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)
Time spent: 26% 30x opt_clean (6 sec), 19% 25x opt_merge (4 sec), ...
OpenROAD 1.1.0 c60f97245d
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  OpenROAD_objects/merged.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  OpenROAD_objects/merged.lef
Startpoint: _34756_ (rising edge-triggered flip-flop clocked by clock)
Endpoint: _36070_ (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
   0.00    0.00   clock network delay (ideal)
   0.00    0.00 ^ _34756_/CK (DFF_X1)
   0.08    0.08 ^ _34756_/Q (DFF_X1)
   0.17    0.26 ^ _30643_/Z (BUF_X1)
   0.02    0.28 v _17612_/ZN (NOR2_X4)
   0.05    0.33 v _17614_/ZN (AND2_X4)
   0.03    0.36 ^ _17703_/ZN (NOR2_X2)
   0.02    0.38 v _17719_/ZN (INV_X1)
   0.05    0.43 ^ _17720_/ZN (AOI221_X2)
   0.04    0.47 ^ _17721_/ZN (AND2_X2)
   0.03    0.50 ^ _17727_/ZN (AND2_X4)
   0.03    0.53 ^ _17749_/ZN (AND2_X4)
   0.03    0.56 ^ _17750_/Z (BUF_X4)
   0.04    0.59 ^ _21064_/ZN (AND3_X4)
   0.04    0.64 ^ _21069_/ZN (AND3_X2)
   0.04    0.67 ^ _21345_/ZN (AND2_X4)
   0.03    0.70 ^ _21451_/Z (BUF_X8)
   0.06    0.76 v _21452_/Z (MUX2_X1)
   0.03    0.79 v _21453_/ZN (AND2_X1)
   0.06    0.85 ^ _21456_/ZN (AOI211_X2)
   0.01    0.86 v _21467_/ZN (NOR3_X1)
   0.04    0.90 v _21487_/ZN (OR2_X1)
   0.06    0.96 v _21488_/Z (MUX2_X1)
   0.03    0.98 v _32559_/Z (BUF_X1)
   0.00    0.98 v _36070_/D (DFF_X1)
           0.98   data arrival time

   1.00    1.00   clock clock (rise edge)
   0.00    1.00   clock network delay (ideal)
   0.00    1.00   clock reconvergence pessimism
           1.00 ^ _36070_/CK (DFF_X1)
  -0.04    0.96   library setup time
           0.96   data required time
---------------------------------------------------------
           0.96   data required time
          -0.98   data arrival time
---------------------------------------------------------
          -0.02   slack (VIOLATED)

Design area 102521 u^2 100% utilization.
[... 353 lines cut ...]
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  OpenROAD_objects/merged.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  OpenROAD_objects/merged.lef
Error: OpenROAD_results/5_route.def does not exist.
Error: cannot open 'OpenROAD_results/5_route.sdc'.
Error: no network has been linked.
Error: no network has been linked.
Error: no network has been linked.
Error: no network has been linked.
Error: no network has been linked.
Error: no network has been linked.
Error: no network has been linked.
Error: Error: no network has been linked.
Error: no network has been linked.
Error: Error: no network has been linked.
Error: no network has been linked.
Error: Error: no network has been linked.
ERROR: /opt/panda/share/panda//scripts/def2gds.py:11: Unable to open file: OpenROAD_results/5_route.def (errno=2) in Layout.read
  /opt/panda/share/panda//scripts/def2gds.py:11 (class RuntimeError)

```

eder-matheus, 2020-05-14:
Hi, @tamimcse. Looks like you are using an old version of FastRoute. Can you try to update your FastRoute submodule with the latest commits of "openroad" branch?

tamimcse, 2020-05-14:
Yeah, I don't see the error in `openroad` branch. However I'm seeing another error. I'm closing the issue and creating a new one.

closed 2020-05-14 by tamimcse