The-OpenROAD-Project-Attic/FastRoute #32: Malformed guides of small 3-node nets?

ax3ghazy, 2020-06-30:
Hello,

This is an issue that appeared since one of the recent commits (i.e., it didn't happen a month ago). When there is a net made up of three nodes, in which two of the nodes are very close to each other, like within the same GCELL (e.g., a block pin connected to a stdcell pin that is connected to diode pin directly next to it), the produced guide seems malformed. The metric here is going through TritonRoute successfully, and in this case it exits, complaining about pins not being covered by the guide.

I have privately shared a test case, in which there are two layouts, one with a 3-node net with 2 nodes close to one another, as described above, and the same layout where one of those two nodes is moved away by a row. The guides for the first case doesn't cover all pins, but those of the second test case go successfully through TritonRoute.

guide1:
```
_021_
(
103499 41399 110399 48299 MET0
103499 41399 117299 48299 MET1
)
```

guide2:
```
_021_
(
110399 41399 117299 48299 MET1
110399 34499 117299 48299 MET2
110399 34499 117299 41399 MET1
110399 34499 117299 41399 MET0
103499 41399 110399 48299 MET0
103499 41399 117299 48299 MET1
110399 41399 117299 48299 MET2
110399 41399 117299 48299 MET1
)
```

Of course the scripts to generate those guide are included in the test case. Please refer to @tspyrou to access the test case.

Thanks!

eder-matheus, 2020-07-06:
@ax3ghazy Can you check the last commit of FastRoute repo (openroad branch)? It should fix the issue that you've reported

ax3ghazy, 2020-07-07:
@eder-matheus: Thank you. It has been resolved.

closed 2020-07-07 by ax3ghazy