The-OpenROAD-Project-Attic/FastRoute #36: Router is assuming all the pins to be in the first layer

mukul19147, 2020-07-23:
Even if I supply pins in the different layer the router is assuming them in the first layer and routing because of which many higher-layer pins remain unconnected. How to fix it?

eder-matheus, 2020-07-24:
@mukul19147 This is a known issue in FastRoute code. We've implemented a post-process workaround that inserts guides for pin access to pins in superior meta layers. Take a look at this function: https://github.com/The-OpenROAD-Project/FastRoute/blob/openroad/src/FastRouteKernel.cpp#L1318

mukul19147, 2020-07-25:
Okay so VIAs are added to cover the uncovered pins. But if we have a constraint on a net that that none of its segments should be below a particular layer than than this router can't be used. How to incorporate this change?