The-OpenROAD-Project-Attic/FastRoute #38: Sink/source verification during a global placement seems to fail on nets connected to pins

rbarzic, 2020-07-23:
When running a global placement on my design, I get an error message (when fastroute is invoked apparently):

Initializing nets...
Checking pin placement...
Checking pin placement... Done!
 > ----Checking sinks/source...
[ERROR] Net CLKIN has invalid sinks/source distribution
    #Sinks: 0; #sources: 1
[ERROR] Net DBG0 has invalid sinks/source distribution
    #Sinks: 0; #sources: 0
[ERROR] Net DBG1 has invalid sinks/source distributio

....

A testcase is available at https://github.com/rbarzic/issues-openroad. directory replace1. It should be self contained
I'm using Openroad, branch openroad  sha1 811dc2ff

lsteveol, 2020-08-05:
@rbarzic Were you able to find any kind of workaround to keep progressing? 

Asking as I'm running into the same issue and just want to flush the flow out.

rbarzic, 2020-08-06:
No. And I didn't try a newer version of OpenRoad either